meshtastic/ATAK-Plugin #15: Show all Meshtastic devices - add old data to the map

daskalovst, 2024-04-22:
The map is displaying Meshtastic devices that are currently inactive. These devices are listed in the nodeDB but haven't been seen for several days, yet they still show up on the map. I think need to be adjusted it so that only active devices are shown. After 900 seconds of inactivity (assuming the default parameter for position updates), the inactive devices should turn black and disappear from the map. or at least to have some indicator that shows when the device are seen for last time

In the remarks section, there is a timestamp, but it's in a format like "1713103054," which isn't readable for humans at all.

It seems that nodeDB isn't a reliable source for this information because it's not functioning correctly. but that's different topic...

caveman99, 2024-04-22:
nodeDB is working fine, but it has a limited capacity.

You proposal sonds good, but i would use the cutoff for inactivity at 2x position update interval. We do that in other parts of the frimware too. (neighborinfo...)

daskalovst, 2024-04-22:
900sek is just assumption based on default parameters, its also good to have neighborinfo in remarks field, this will help determinate real coverage of your own device.

nodeDB doesn't work adequate even for less than 10devices. but i don't know where to report it.

niccellular, 2024-04-24:
The sensor nodes have a stale time of 10 minutes, they should disappear if they are not seen after that. The plugin doesn't know anything about nodeDB, it simply listens for "com.geeksville.mesh.NODE_CHANGE" events from IMeshService and generates a CoT if they have none zero gps coords.

daskalovst, 2024-04-26:
I have no idea how com.geeksville.mesh.NODE_CHANGE is generated from the meshtastic app, but on the ATAK map appears, nodes that are not active from days even mounts, and stay on the map, if there have stale time 10min they should disappear from the map, in this case somehow meshtastic app continue generating com.geeksville.mesh.NODE_CHANGE massage, to update those nodes in ATAK map

closed 2024-07-29 by niccellular