meshtastic/ATAK-Plugin #16: [Question] Green icon with no hardware attached

daskalovst, 2024-04-22:
The meshtastic icon in ATAK is green if no hardware connected to the meshtastic app.
Is there have plan this to be corrected in the future versions?

ATAK version 5.1.0.6
Plugin version 1.0.11
meshatstic app 2.3.3

daskalovst, 2024-04-22:
It appears this isn't a question, but rather an observation about an issue with the green icon in ATAK even when no hardware is attached. I've attempted to recreate the issue several times, but with little success. Only once out of around 15 attempts did the icon stay green. This is happen when the Meshtastic device was out of range of Bluetooth. This intermittent functionality of the feature is puzzling, as it's difficult to understand what causes it.

daskalovst, 2024-04-26:
Yes, I've noticed that most of the time, the icon color changes correctly. I've been testing it for 4 days now, and during this period, there were 3 instances where the icon stayed green despite the hardware not being connected. All of these occurrences happened when the hardware was outside the Bluetooth range. However, this issue is quite rare, and it seems there might be an additional factor triggering this malfunction. I'm still trying to determine what exactly is causing it.

niccellular, 2025-04-14:
The plugin doesn't really know about hardware, it simply talks to the Android app's service.

closed 2025-08-30 by niccellular