meshtastic/ATAK-Plugin #18: [Question] Rationale to revert back to previous file naming convention?

mtsdawson, 2024-04-24:
The 4.10.0 in the filename has created some confusion with users as it suggests a unique fork for the legacy ATAK version.  This appeared to be resolved with v1.0.11 as shows v5.0.0 in the filename.

What's the rationale to revert back to the v4.1 designation in the filename? Any chance this can be removed entirely to reflect cross-version compatibility? 

Filenames:
ATAK-Plugin-Meshtastic.Plugin-1.0.12-d77f5b26-**4.10.0**-civ-release.apk
ATAK-Plugin-Meshtastic.Plugin-1.0.11-4ac447bb-**5.0.0**-civ-release.apk

Thanks for the great work on this plugin.

niccellular, 2024-04-24:
when you submit the source code to TPC for building, they name the file as plugin version, git hash, atak version.

prior to atak 4.10, plugins had to be recompiled for the particular atak version.

4.10+ plugins can now run on 4.10, 5.0, 5.1 etc

mtsdawson, 2024-04-24:
Correct, therefore would it be possible to either A) remove 4.10 from the filename, or B) change it to 5.0.0 as you did in the previous build?

I get a lot of noise from users hesitant to install this plugin when it lists v4.10 in the APK filename.

niccellular, 2024-04-24:
tell them 4.10 plugins work on 5.x

closed 2024-04-24 by niccellular