meshtastic/ATAK-Plugin #21: T-Beam keeps restarting after lauchning ATAK with Meshtastic plugin.

RybczynskiM, 2024-05-17:
ATAK: 5.1.0.12
Meshtastic fw: 2.3.9 alfa
Device: LILYGO T-Beam Supreme 433MHz Ublox GPS.

After launching ATAK, device keeps rebooting (resetting). Also, when using user button to switch screens they keep going back to the first screen after a sec. Tested this on Beta and Alfa firmware.

Does the new releases work with newer ATAK versions (i.e. 5+) and what is the current supported Meshtastic firmware?

caveman99, 2024-05-17:
the ATAK Plugin is rather firmware agnostic. Can you provide a serial log of the t-beam when it is rebooting?

niccellular, 2024-05-18:
Are you using the 1.0.21 ATAK plugin? And yes all 4.10.0 plugins work on ATAK 4.9+

RybczynskiM, 2024-05-21:
Hey. Sorry for late response. Im testing T-Beam Supreme only. No connection to other devices no more recivers, only one phone and one T-Beam running TAK role.

Here is log file from console connection. 
20:04:03 is the moment I placed my position on map on ATAK app.

[meshtastic-log-2024-05-21T20-06-42.323Z.zip](https://github.com/meshtastic/ATAK-Plugin/files/15395006/meshtastic-log-2024-05-21T20-06-42.323Z.zip)

niccellular, 2024-05-21:
I don't see any crash in the log?

niccellular, 2024-07-29:
This was probably fixed with the uni2shox stuff, see 2.4.x firmwares

closed 2024-07-29 by niccellular