meshtastic/ATAK-Plugin #22: Excessive Channel Utilization

DB-SOC, 2024-05-20:
While the addition of "Only send PLI and Chat messages over Meshtastic" option has significantly reduced the volume of traffic forwarded to the mesh, there is still an excessive amount of data being forwarded. Far more than the 5 minute continuous PLI updates I have configured on both devices, and persists when no chat traffic is being sent. Approximately every 15 seconds both nodes broadcast. This causes a constant 15-25% channel utilization on the Long_Fast modem preset.

5.1.0.7 ATAK
1.0.21 Meshtastic Plugin
2.3.3 Meshtastic App
2.3.6 Meshtastic Firmware on Heltec V3 and CanaryOne

niccellular, 2024-05-21:
My only suggestion is to use Med/Fast

niccellular, 2024-07-29:
Double check you didn't enable multiple ATAK plugins with enable relay to server

closed 2024-07-29 by niccellular