meshtastic/ATAK-Plugin #24: [Feature Request] Team colors for tracker nodes

habr05, 2024-05-26:
Presently, only the Sensors are supported by team color tag. Is there anything that prevents it to include Tak Tracker nodes too?

niccellular, 2024-05-29:
I think this should work already? Just append the node color scheme to the end of the long name.

habr05, 2024-05-29:
it works for every type except "TAK TRACKER"

habr05, 2024-05-29:
Both, "CTmA 6924 -5" and "TARDY", are the same node.

[image]

niccellular, 2024-06-05:
I just tested this and it's working for me?

niccellular, 2024-06-07:
Make sure you have "Show all Meshtastic devices" enabled

habr05, 2024-06-07:
So you're saying that team color, for TAK_TRACKER node type, will show only when "Show all Meshtastic devices" is enabled. Is that correct?

niccellular, 2024-06-09:
> So you're saying that team color, for TAK_TRACKER node type, will show only when "Show all Meshtastic devices" is enabled. Is that correct?

Yes

habr05, 2024-06-13:
What's the reasoning behind it? Why not keep it consistent regardless of the "Show all Meshtastic devices" settings?

niccellular, 2024-06-25:
tracker nodes are meshtastic devices

habr05, 2024-06-25:
ATAK has a standalone "tracking only" app used for anything that does not have (require) user input. TAK_TRACKER would be equivalent of that app. In real life that would be useful to slap those trackers on equipment, team members or even as beacons for  landing pad, medic, command post etc. 

Just to make sure that we are talking about the same thing I'm including the screenshot of the selected option.

[image]

niccellular, 2024-06-25:
it is not equiv

habr05, 2024-06-25:
the problem is that when I enable "show all..." the team color is applied to the node BUT if I disable "show all..." it is not, and the appended -# shows in the name.

niccellular, 2024-06-29:
try using the "TRACKER" role not "TAK_TRACKER" i probably have a bug somewhere if what you are saying is correct. I'll try to fix it soon, but if you can try just using TRACKER role maybe it will get you going for now

alexsh5, 2024-07-08:
This is the issue I have experienced for some time. You're right that it only affects the TAK TRACKER role. The only thing I can find that might relate is the statement before checking colours in the long name. Checks if "plugin_meshtastic_tracker" is FALSE. Does this mean it is intentionally excluding the TAK TRACKER role?

[https://github.com/meshtastic/ATAK-Plugin/blob/e0423ac2e826601a9d9781895e12c847b7427921/app/src/main/java/com/atakmap/android/meshtastic/MeshtasticReceiver.java#L354](url)

niccellular, 2024-07-29:
Its getting the value of `plugin_meshtastic_tracker` if it is not set then the default is `false`

To use the role-by-name thing, you must have that preference enabled

RichieFrame, 2024-09-20:
For context, the reason that this does not work on TAK TRACKER is because the information arrives from those nodes in a tak packet format and already includes a team color, which is hardcoded in the firmware. I put in a feature request for this to be added to the firmware

RichieFrame, 2024-09-23:
https://github.com/meshtastic/firmware/discussions/4787
I guess we vote on feature requests now? If you want this, vote for it at the above link

habr05, 2024-09-25:
> [meshtastic/firmware#4787](https://github.com/meshtastic/firmware/discussions/4787) I guess we vote on feature requests now? If you want this, vote for it at the above link

voted

closed 2025-08-30 by niccellular

RichieFrame, 2025-09-25:
I have test firmware running on my t-echo which lets me set the team color on demand from the device UI, which is one step closer to making the color a device config option so it can survive a reboot/shutdown, but its usable right now

I also added a build time override for the default color, maybe I can get this into the firmware with a little more work

RichieFrame, 2025-12-16:
The firmware repo is accumulating a large number of unresolved issues and unmerged PRs, I do not want to drop a whole bunch of code in there and have it get stale and umergeable, also at some point a UI change broke my menus and I gotta fix that 🤦 

My local branch is mostly caught up but the color codes were not the only thing in there, I had a lot of custom code for t-echos (fixes to be upstreamed) and custom algorithm for determining GPS update intervals (just for me) that I need to deal with as well, once I fix the menus I will try to get some of the t-echo code upstreamed, then back to this!

I can also clone the repo with changes to github so y'all can build that firmware for your own devices at will, if that will be helpful.

niccellular, 2026-02-08:
This is a firmware thing, @RichieFrame needs to submit his code to firmware people