meshtastic/ATAK-Plugin #27: Playstore Report - no additional information

bisgroves, 2024-06-10:
java.lang.RuntimeException
Exception java.lang.RuntimeException:
  at android.app.LoadedApk$ReceiverDispatcher$Args.lambda$getRunnable$0 (LoadedApk.java:1919)
  at android.app.LoadedApk$ReceiverDispatcher$Args.$r8$lambda$gDuJqgxY6Zb-ifyeubKeivTLAwk
  at android.app.LoadedApk$ReceiverDispatcher$Args$$ExternalSyntheticLambda0.run
  at android.os.Handler.handleCallback (Handler.java:958)
  at android.os.Handler.dispatchMessage (Handler.java:99)
  at android.os.Looper.loopOnce (Looper.java:230)
  at android.os.Looper.loop (Looper.java:319)
  at android.app.ActivityThread.main (ActivityThread.java:8919)
  at java.lang.reflect.Method.invoke
  at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run (RuntimeInit.java:578)
  at com.android.internal.os.ZygoteInit.main (ZygoteInit.java:1103)
Caused by java.lang.NullPointerException:
  at com.atakmap.android.meshtastic.MeshtasticMapComponent.getMyNodeID
  at com.atakmap.android.meshtastic.MeshtasticReceiver.onReceive
  at android.app.LoadedApk$ReceiverDispatcher$Args.lambda$getRunnable$0 (LoadedApk.java:1911)

niccellular, 2024-06-10:
Hi, what version of the plugin was this on? The code around the call is in a try,catch and the return value is checked for null.

bisgroves, 2024-06-10:
> External Email: This email has originated from outside of the organization. Only click links or open attachments if you are certain the contents are safe.

Unsure.  If this is now addressed, feel free to close this out

closed 2024-06-10 by niccellular