meshtastic/ATAK-Plugin #29: [Feature Request] Enable Relay to Meshtastic

bmswens, 2024-07-18:
As a user, when I receive data from a TAK server (that isn't also received via Meshtastic), I would like the option to automatically forward that data to the mesh.

I think it would be most intuitive to be added as another preference alongside the "Enable relay to server". It's the same idea, just reversed data flow.

Use case: 

I may have multiple users in the field running ATAK with the Meshtastic Plugin, a single user connected to both the mesh and a local TAK server, and several "HQ" personnel utilizing WinTAK, WebTAK, etc. 

As of right now, all of the TAK server users will receive data from the Meshtastic users in the field, but any data added by the TAK server users will have to be manually broadcasted by the one node that is connected to both.

niccellular, 2024-07-29:
No, this will flood your meshtastic network.

bmswens, 2024-07-30:
Would you consider at least supporting minimal COT events, such as the ones that are already supported via the plugin? 
Even just PLI and GeoChat events would help integration immensely.

niccellular, 2024-08-05:
Takserver doesn't speak the meshtastic protobufs

niccellular, 2024-08-06:
@bmswens See release 1.0.28