meshtastic/ATAK-Plugin #31: Sending routes with more than 2 points is very unreliable

aramyan, 2024-08-27:
Setup: 2 android tablets with ATAK and official ATAK Meshtastic plugin (latest versions, 5.2.0.4 and 1.0.30), 2 T-Beam supremes with latest stable firmware (2.4.2) and a TinySA ultra to monitor the frequency, indoor, everything laid down on the same table, WiFi and MQTT disabled.
LoRa configuration: custom - 250 bandwidth, 8 spread factor, 8 coding rate, EU_868 (best results with these settings).

Chat in ATAK works most of the time.
Sending single markers in ATAK works most of the time.
Sending routes made up of 2 points works most of the time.
Sending routes made up of 3 points or more gets very unreliable, and unreliability increases the more points you add.
Sending hand drawn lines on the map is basically non-functional, you can maybe get one through after dozens of attempts but that’s it.

Signal strength between nodes as reported by Meshtastic app is at -15 dBm and about 12.0 SNR so clearly radios have no problem hearing each other.
I draw a path of many points or a few hand drawn lines in ATAK, I hit send, I see the transmission start and end on tinySA, and then nothing appears in ATAK on the other tablet. If I keep trying maybe after 10-20-30 times it will get through.

Please help.

niccellular, 2024-08-27:
Can you try on Med/Fast? No one is doing any tests with custom LoRa configs...

aramyan, 2024-08-27:
So I went ahead and tested with medium fast and then again with custom settings. Medium fast is Bandwidth: 250, Spread Factor: 9, Coding Rate: 5 (according to this page https://meshtastic.org/docs/overview/radio-settings/), and my custom is 250/8/8, so shorter spread factor but more coding rate (100% redundant data instead of 25%).

I tested sending a 2 point path, a 3 point path, a 5 point path and a 17 point path, 10 times each. The results (of how many times out of 10 the other tablet got the path and showed it on the map) were as follows.

Medium Fast (250/9/5)
2 point path: 4/10
3 point path: 5/10
5 point path: 3/10
17 point path: 1/10

Custom (250/8/8)
2 point path: 6/10
3 point path: 5/10
5 point path: 6/10
17 point path: 1/10

So slightly better success rate with more coding rate, but still very random and inconsistent.

I then enabled the logcat in ATAK, relaunched ATAK and sent the 17 point path maybe a dozen times and got 3 successes, two in the beginning and one at the end. I checked the debug panel of MT app and there were plenty of ATAK_FORWARDER portnums. I also checked the logcat and there were some errors there, but I can't make sense of it. I'm attaching it in case it helps someone who understand it better make sense of it.
[logcat0.txt](https://github.com/user-attachments/files/16768498/logcat0.txt)

niccellular, 2024-08-28:
A 17 point path is like 4kB, that will be a stretch to send without a single error. Could try short/fast or the new short/turbo. The problem is how long it takes to TX something that big. I'd be surprised if you weren't hitting your duty cycle limits.

aramyan, 2024-08-28:
And yet even 2-3-5 point paths don't get through half the time (not that getting 4kB of data transferred intact over radio should be a problem in 2024). The problem that is relevant to this discussion is that there is apparently no mechanism to make sure data payload gets to its destination intact even over a perfectly good connection. If this is intended behavior, then this should be clearly stated somewhere, that ATAK+MT+Official plugin setup does not in any way protect the integrity of your data beyond coding rate settings, which are set to lowest of 5 on most modem presets, which in turn means you'll be failing to send anything other than a marker or a chat message 40-60% of the time. Software that allowed me to download 600 MB files intact over shoddy WiFi existed in 2001, there's no reason this software can't handle 4kB of data, and yet the fact that it can't is not clearly expressed anywhere, and you're here addressing my duty cycle limit instead of the very significant fact that the setup advertised as *the* solution to offgrid comms is borderline dysfunctional and half the time can't transfer a rectangle drawn on a map.

niccellular, 2024-08-28:
Another option which might be more robust is to send it as a datapackage. The logic in the plugin for sending a datapackage attempts to wait for ACKs before firing off the next chunk of a payload.

GUVWAF, 2024-08-28:
> Software that allowed me to download 600 MB files intact over shoddy WiFi existed in 2001, there's no reason this software can't handle 4kB of data, and yet the fact that it can't is not clearly expressed anywhere

Then you were probably using TCP over Wi-Fi as e.g. UDP is still "best-effort", just like Meshtastic. You can read about its limitations [here](https://meshtastic.org/docs/overview/mesh-algo/#layer-1-unreliable-zero-hop-messaging). TCP adds a lot of overhead which is simply unnecessary for most use-cases of Meshtastic. If you're relying on multiple packets being sent correctly in a row, there is no guarantee this will work.

@niccellular Does the plugin rely on packets being received in order or are there sequence numbers added such that you can reshuffle them?

aramyan, 2024-08-28:
> TCP adds a lot of overhead which is simply unnecessary for most use-cases of Meshtastic.

While I certainly don't have statistical data to know what use-cases of Meshtastic are prevalent in general, but you would at least agree that ATAK + Meshtastic setup appeals primarily to outdoor/wilderness users? Now imagine if someone or a group of people goes hiking into the woods and encounters a dangerous situation and then tries to communicate with outside world using ATAK+MT setup, not knowing that data transfer is never 100% reliable and gets increasingly unreliable as you try to use anything other than lone markers. There is potential for someone to come to harm because of this.

So while I understand that this is a free open-source project and that devs don't *have* to do anything they don't want to, I at least hope that this state of affairs is communicated to potential users. Yes, ATAK is a piece of software that was developed with broadband connection in mind. Yes, Meshtastic is a piece of software that was written with no data integrity protection other than some redundant overhead (which is set to minimum in majority of modem presets), because it's intended to be used with short messages. But their pairing exists through official plugin, and sounds very exciting, until you spend time and resources getting the hardware, setting it up, and only then learning that hey, data transfer over MT can get corrupted pretty easily even with good radio link, and there's nothing either ATAK or MT will do about it, and you better be in a safe environment when you learn that significant detail, because if you're not, you might be quite out of luck.

So if devs can implement some data integrity protection in the future, that would be great and would make the setup as exciting as it sounds on paper. If not, I ask that it is at least clearly stated that data transfer is never 100% reliable, and gets increasingly unreliable with larger payloads. I feel that this is important to a large percentage of users of this setup, to a point when under some circumstances their physical wellbeing might depend on it.

niccellular, 2024-08-28:
> @niccellular Does the plugin rely on packets being received in order or are there sequence numbers added such that you can reshuffle them?

Currently, it needs the packets to arrive in order, unless you send a data package because that code path waits for an ACK for each packet chunk.

GUVWAF, 2024-08-28:
> Currently, it needs the packets to arrive in order

In that case, @aramyan, you might have been impacted by this bug (https://github.com/meshtastic/firmware/issues/4560), which slipped in in firmware 2.4.1 and was found and fixed two days ago, but there’s no new release yet. It caused packets to go out of order **already in the transmitter’s queue**. You could try to temporarily roll-back to 2.4.0 and see if things improve.

> While I certainly don't have statistical data to know what use-cases of Meshtastic are prevalent in general

I also don't have the numbers, but I'm pretty certain the percentage of users of Meshtastic that is using it in combination with ATAK is very limited, among others because it requires Android.

> and there's nothing either ATAK or MT will do about it, and you better be in a safe environment when you learn that significant detail, because if you're not, you might be quite out of luck.

To be clear, it would be very nice if it *did* work reliably all the time and I'm trying to figure out what the main causes are. However, you seem to be uninformed about Meshtastic if you're saying "_there's no reason this software can't handle 4kB of data_". Because, as you can read in the documentation, there is only 237 bytes of payload per packet, so you need at least 17 packets to be sent, but there is nothing in Meshtastic's protocol to ensure that multiple packets are received correctly in a row, let alone in order. If you feel this is not clear, please open an issue for the documentation or improve it yourself.

niccellular, 2024-08-28:
> @niccellular Sorry to bother you, but now I'm curious. Might it be the plugin is not handling [`queueStatus`](https://github.com/meshtastic/protobufs/blob/431291e673c1c7592ee64cb972d7845589f60138/meshtastic/mesh.proto#L1476) packets from the device like the [Android app](https://github.com/meshtastic/Meshtastic-Android/blob/b4cdbf0617eb49df24c459d5d172a6e3ab48e091/app/src/main/java/com/geeksville/mesh/service/MeshService.kt#L1307) does? LoRa is pretty slow and there are only 16 entries in the device's Tx queue, so if you're sending it too fast, if might be full and then it simply drops the packet.

Not sure if that is exposed in the AIDL from the MT android app. The only API into it I am doing is calling `send`

GUVWAF, 2024-08-28:
> Not sure if that is exposed in the AIDL from the MT android app. 

From a second look it looks indeed this is incorporated already in the `MeshService` which the plugin is using, so that's good.

niccellular, 2024-08-28:
@aramyan Please try the experiment again using datapackages. The logic for those should improve the reliability and hopefully get you the results you want.

aramyan, 2024-08-28:
> @aramyan Please try the experiment again using datapackages. The logic for those should improve the reliability and hopefully get you the results you want.

Thanks, I'll try that.

niccellular, 2024-08-28:
> > @aramyan Please try the experiment again using datapackages. The logic for those should improve the reliability and hopefully get you the results you want.
> 
> Thanks, I'll try that.

Sorry for it to work you must enable a preference "Allow SWITCH command" which will reboot your nodes into short/fast but will return them to whatever they are set to after, hopefully...

aramyan, 2024-08-28:
> Sorry for it to work you must enable a preference "Allow SWITCH command" which will reboot your nodes into short/fast but will return them to whatever they are set to after, hopefully...

Yeah that just happened and I got startled and was about to ask if I can stop the nodes from rebooting to short/fast, but it does go through more reliably. Also, since I'm not choosing any contacts to send to, does it mean it will send to all nodes that are running ATAK and have switch command enabled?

niccellular, 2024-08-28:
> Yeah that just happened and I got startled and was about to ask if I can stop the nodes from rebooting to short/fast, but it does go through more reliably. Also, since I'm not choosing any contacts [...]

Yes, its a OPT-IN thing as to prevent other people on your mesh from forcing you to switch.

aramyan, 2024-08-30:
> Yes, its a OPT-IN thing as to prevent other people on your mesh from forcing you to switch.

Thanks, I'll do some outdoor tests today. Hope proper integrity features come in the future. Until then we'll get by with what we have.

closed 2024-08-30 by aramyan

InconsolableCellist, 2024-08-31:
Is this really closed as completed? I'm following with interest, as I also would like to see either a guarantee of reliability when attempting to transmit critical information (and basically any communication in ATAK should be considered mission critical), or enough information about the transmission that you know to retry. Or perhaps can set an automatic retry limit to maximize the chances of it getting through.

A feeling of unreliability is like a disease, where if the user perceives it in your application, they'll never be able to feel they can fully trust it. Obviously data transmission and integrity here is dependent on Meshtastic, but everything that's reasonable should be done to inform the user as to what's making it through or not, and what they can do instead in the case of an error (like continual retransmission or requests for ACK).

I'm willing to bet without even checking that military radios as the backend for ATAK have all these features and more, because you simply can't have an application for sending mission critical information where it's a shrug and a best-guess as to whether it's going to get through, especially in ideal conditions.

niccellular, 2024-09-01:
See build 1.0.31, I reworked the chunked messages so they are ACK'd just like a data package. Should improve reliability without having to use the data package workaround.

aramyan, 2024-09-01:
> See build 1.0.31, I reworked the chunked messages so they are ACK'd just like a data package. Should improve reliability without having to use the data package workaround.

I didn't expect this, assuming you were too busy to commit to implementing this, but big thanks to you for doing so. I'm sure that usage of ATAK with MT will increase in the future, and maybe MT will implement reliable delivery all the way (not just between immediate neighbors), because ATAK+MT is just too useful of a combo, especially for communities with not a lot of infrastructure to rely on. Thanks again <3