meshtastic/ATAK-Plugin #32: TAK nodes not seen by eachother

RichieFrame, 2024-08-29:
I have 2 devices (wisblock) in TAK mode, both connected to Android phones running the latest ATAK, latest official plugin, and latest Meshtastic app.
I have 2 devices (t-echo) running in TAK_TRACKER mode.
All 4 devices are running 2.4.3 firmware.
When all devices were configured as client, the app saw the position updates for all trackers, the trackers saw position updates for all clients, and all messaging was functional.

In ATAK, I can see the tracker nodes, the positions update every 3 mins as configured, though they contain no altitude or DOP data.
The tak nodes however cannot see eachother, and sent messages in ATAK chat are also not received on the other node.
If I reconfigure the tak node with GPS as TAK_TRACKER it can be seen by the other tak node.
If I send a message using the app from a tak node, it is received by all other nodes.

This leads me to believe the following:
Meshtastic mesh is stil working
ATAK is communicating with the plugin
The plugin is communicating with the app
Something is wrong in the chain, preventing TAK packets from ATAK getting to the mesh, or preventing received TAK packets from being accepted by other nodes

By the issues list this does not appear to be a known issue, if there is some kind of setting I am missing in ATAK to have it send I do not know what it would be. I can attempt to make changes, test pre-release plugins, and get debug logs if any of that will help.

niccellular, 2024-08-30:
try the "TRACKER" role instead of "TAK_TRACKER"

RichieFrame, 2024-09-15:
Tried TRACKER, and TAK_TRACKER, behaved the same as TAK in that they show the standalone t-echos on the maps but did not show the ATAK device on the other ATAK node, and global chat is not received by the other node. In TAK_TRACKER the meshtastic device's long name shows up on the other ATAK map/contact list, as would be expected because it is broadcasting its own PLI packets independent of ATAK.

RichieFrame, 2024-09-17:
I just installed the latest plugin version and meshtastic app on both devices and they can now see eachother!! I also have functional chat via the global geochat room but not user to user

RichieFrame, 2024-09-17:
After messing around a bit, I saw it was throwing an error with a local LAN IP when trying to direct message, I had started up ATAK initially with both devices on WIFI, then did the mesh tests with it turned off on one device.... somehow it had pinned its IP as the message target, restarting ATAK is now allowing direct messages to be sent in addition to global, and sending map pins is also working albeit slowly (30s)

RichieFrame, 2024-09-20:
Results of further testing: only one device can receive direct messages from the other, I do see some kind of odd error in the ATAK logs of the sending device, it looks like one of the uids is getting truncated, and it causes an exception. The receiving node gets the packet with a recipient uid that does not match, I will get some more details from these logs and post it.

But before I do that, it looks like the truncation is happening in the uid where there are consecutive 0s, like this:
ANDROID-e7e455b40002429d turns into ANDROID-e7e455b4 on the other device, so the 00 might be turning into a null char somewhere and borking the packet

Luthix, 2024-10-03:
In the latest version, I'll get duplicate nodes of the same device. When I go to the contact info one of them has no Version line and in Node Type will just say Unit > Combat and that one can't be messaged. The other under Version will say Meshtastic Plugin and then all the NodeInfo and its Node Type is Equipment > Sensor can receive messages through the Atak app. I can replicate this by quitting and restarting the Meshtastic app and doing so will place nodes of every device that has ever connected to that mesh even if they are offline and the device that is online will get this duplicate behavior.

davidmcginnis3, 2024-11-18:
v4.10.0.56 ver ATAK
Meshtastic plugin v1.0.35
Meshtastic app 2.5.7

I am seeing the same behavior in respect to the IP address following the wifi connection.  Of course the eud are not used where wifi is always available.  Work around is start the atak and meshtastic app with wifi disabled.

Dima-Pima, 2025-01-19:
Meshtastic node_1 with role TAK. This node is connected to phone with ATAK app.
Meshtastic node_2 with role Client. This node is connected to iPhone with Meshtastc app.

Messages from node_2 is displayed in ATAK app, but in chat with node_1 LongName.
Answers in this chat from node_1 to node_2 are not delivered.

ATAK v5.3.0.9
Plugin v1.0.36

closed 2025-08-30 by niccellular