meshtastic/ATAK-Plugin #36: [ERROR] Reboot Lilygo T3 1.6.1 2021

dominguez255, 2024-10-19:
The card restarts by itself every 30 seconds or when sending a message through ATAK; however, when I don't use ATAK and use the Meshtastic app, it works perfectly

ATAK 4.10.0.42
ATAK Plugin Meshtastic v1.0.35
Lilygo T3 1.6.1 2021 with firmware tlora-v2-1-1p6 2.4.2.5b45303

INFO | ??:??:?? 26 Received atak from=0x0, id=0x6c1a0e45, portnum=72, payloadlen=56
DEBUG | ??:??:?? 26 Received uncompressed TAK payload from phone: 56 bytes
DEBUG | ??:??:?? 26 Compressed callsign: 5 bytes
DEBUG | ??:??:?? 26 Compressed device_callsign: 17 bytes
DEBUG | ??:??:?? 26 Compressed chat message: 3 bytes
DEBUG | ??:??:?? 26 Compressed chat to: 14 bytes
DEBUG | ??:??:?? 26 Final payload: 53 bytes
DEBUG | ??:??:?? 26 Module 'atak' considered
DEBUG | ??:??:?? 26 Module 'routing' wantsPacket=1
INFO | ??:??:?? 26 Received routing from=0x0, id=0x6c1a0e45, portnum=72, payloadlen=53
DEBUG | ??:??:?? 26 Routing sniffing (id=0x6c1a0e45 fr=0x00 to=0xff, WantAck=1, HopLim=3 Ch=0x0 Portnum=72)
DEBUG | ??:??:?? 26 Module 'routing' considered
DEBUG | ??:??:?? 26 localSend to channel 0
DEBUG | ??:??:?? 26 [Screen] (bw=250, sf=11, cr=4/5) packet symLen=8 ms, payloadSize=73, time 780 ms
DEBUG | ??:??:?? 26 Setting next retransmission in 8244 msecs:  (id=0x6c1a0e45 fr=0x00 to=0xff, WantAck=1, HopLim=3 Ch=0x0 Portnum=72)
DEBUG | ??:??:?? 26 Add packet record (id=0x6c1a0e45 fr=0x00 to=0xff, WantAck=1, HopLim=3 Ch=0x0 Portnum=72)
DEBUG | ??:??:?? 26 Expanding short PSK #1
DEBUG | ??:??:?? 26 Using AES128 key!
DEBUG | ??:??:?? 26 ESP32 crypt fr=2fa69890, num=6c1a0e45, numBytes=57!
DEBUG | ??:??:?? 26 enqueuing for send (id=0x6c1a0e45 fr=0x90 to=0xff, WantAck=0, HopLim=3 Ch=0x8 encrypted hopStart=3)
DEBUG | ??:??:?? 26 txGood=1,rxGood=1,rxBad=0
INFO | ??:??:?? 26 Telling client we have new packets 6
INFO | ??:??:?? 26 BLE notify fromNum
INFO | ??:??:?? 26 getFromRadio=STATE_SEND_PACKETS
DEBUG | ??:??:?? 26 [RadioIf] Starting low level send (id=0x6c1a0e45 fr=0x90 to=0xff, WantAck=0, HopLim=3 Ch=0x8 encrypted hopStart=3 priority=64)
DEBUG | ??:??:?? 26 [RadioIf] (bw=250, sf=11, cr=4/5) packet symLen=8 ms, payloadSize=73, time 780 ms
DEBUG | ??:??:?? 26 [RadioIf] AirTime - Packet transmitted : 780ms
DEBUG | ??:??:?? 27 [RadioIf] Completed sending (id=0x6c1a0e45 fr=0x90 to=0xff, WantAck=0, HopLim=3 Ch=0x8 encrypted hopStart=3 priority=64)
INFO | ??:??:?? 29 To Radio onwrite
DEBUG | ??:??:?? 29 PACKET FROM PHONE (id=0x6c1a0e46 fr=0x00 to=0xff, WantAck=1, HopLim=3 Ch=0x0 Portnum=72)

DEBUG | ??:??:?? 29 handleReceived(USER) (id=0x6c1a0e46 fr=0x00 to=0xff, WantAck=1, HopLim=3 Ch=0x0 Portnum=72)

DEBUG | ??:??:?? 29 Module 'atak' wantsPacket=1

INFO | ??:??:?? 29 Received atak from=0x0, id=0x6c1a0e46, portnum=72, payloadlen=62

DEBUG | ??:??:?? 29 Received uncompressed TAK payload from phone: 62 bytes

Guru Meditation Error: Core 0 panic'ed (Unhandled debug exception).

Debug exception reason: Stack canary watchpoint triggered (nimble_host)

Core 0 register dump:

PC : 0x40085b69 PS : 0x00060a36 A0 : 0x00060a30 A1 : 0x3ffe6b50

A2 : 0x00000000 A3 : 0x00000000 A4 : 0x3ffe6e20 A5 : 0x00000000

A6 : 0x00000000 A7 : 0x00000000 A8 : 0x00000000 A9 : 0x3ffe6c20

A10 : 0x3ffe6c20 A11 : 0x3f401716 A12 : 0x000000ff A13 : 0x0000ff00

A14 : 0x00000005 A15 : 0xff000000 SAR : 0x0000000a EXCCAUSE: 0x00000001

EXCVADDR: 0x00000000 LBEG : 0x40094485 LEND : 0x40094495 LCOUNT : 0xffffffff

Backtrace: 0x40085b66:0x3ffe6b50 0x00060a2d:0x3ffe6f20 |<-CORRUPTED

ELF file SHA256: 3309f9e6999b2478

E (33504) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0

Rebooting...

ets Jun 8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

configsip: 188777542, SPIWP:0xee

clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00

mode:DIO, clock div:2

load:0x3fff0030,len:1184

load:0x40078000,len:13232

load:0x40080400,len:3028

entry 0x400805e4

dominguez255, 2024-10-19:
@thebentern

GMaska, 2024-10-22:
I am having the same issue:
 DEBUG Guru Meditation Error: Core 0 panic'ed (Unhandled debug exception).

Debug exception reason: Stack canary watchpoint triggered (nimble_host)

Core 0 register dump:

PC : 0x40085b69 PS : 0x00060036 A0 : 0x00060030 A1 : 0x3ffe7cf0

A2 : 0x0000002f A3 : 0x00000000 A4 : 0x0000000a A5 : 0x0000001c

A6 : 0x00000064 A7 : 0x3ffe8004 A8 : 0x0000000a A9 : 0x00000000

A10 : 0x00000000 A11 : 0x0000002f A12 : 0x00000000 A13 : 0x3ffe8350

A14 : 0x3ffe8300 A15 : 0x0000000c SAR : 0x0000001c EXCCAUSE: 0x00000001

EXCVADDR: 0x00000000 LBEG : 0x40094485 LEND : 0x40094495 LCOUNT : 0xffffffff

Backtrace: 0x40085b66:0x3ffe7cf0 0x0006002d:0x3ffe7dd0 |<-CORRUPTED

ELF file SHA256: f9eb6950e0627363

E (9675) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0

Rebooting...

ets Jun 8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

configsip: 188777542, SPIWP:0xee

clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00

mode:DIO, clock div:2

load:0x3fff0030,len:1184

load:0x40078000,len:13232

load:0x40080400,len:3028

entry 0x400805e4

E (830) esp_core_dump_flash: No core dump��ѥѥ���found!

E (830) esp_core_dump_flash: No core dump partition found!

[ 11][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz

��@INFO | ??:??:?? 0 

//\ E S H T /\ S T / C

INFO | ??:??:?? 0 Booted, wake cause 0 (boot count 1), reset_reason=reset

niccellular, 2024-11-10:
Upgrade to the latest alpha, they added more memory allocation for the esp32 devices.

closed 2024-11-10 by niccellular