meshtastic/ATAK-Plugin #37: Unable to compile plugin for 5.x

ernest-ibai, 2024-11-14:
Steps:
1. Download zip archive of ATAK-Plugin
2. Open https://tak.gov/user_builds
3. Upload zip file

ER: Build successful
AR: Compile errors:

```
> Task :app:compileCivReleaseKotlin
e: file:///app/plugin-src/ATAK-Plugin-1.0.35/app/src/main/java/com/geeksville/mesh/DataPacket.kt:57:24 Unresolved reference: Portnums
e: file:///app/plugin-src/ATAK-Plugin-1.0.35/app/src/main/java/com/geeksville/mesh/DataPacket.kt:65:33 Unresolved reference: Portnums
e: file:///app/plugin-src/ATAK-Plugin-1.0.35/app/src/main/java/com/geeksville/mesh/DataPacket.kt:70:54 Unresolved reference: MeshProtos
e: file:///app/plugin-src/ATAK-Plugin-1.0.35/app/src/main/java/com/geeksville/mesh/DataPacket.kt:73:24 Unresolved reference: Portnums
e: file:///app/plugin-src/ATAK-Plugin-1.0.35/app/src/main/java/com/geeksville/mesh/DataPacket.kt:77:19 Unresolved reference: MeshProtos
e: file:///app/plugin-src/ATAK-Plugin-1.0.35/app/src/main/java/com/geeksville/mesh/DataPacket.kt:78:33 Unresolved reference: Portnums
e: file:///app/plugin-src/ATAK-Plugin-1.0.35/app/src/main/java/com/geeksville/mesh/DataPacket.kt:79:13 Unresolved reference: MeshProtos
e: file:///app/plugin-src/ATAK-Plugin-1.0.35/app/src/main/java/com/geeksville/mesh/NodeInfo.kt:20:22 Unresolved reference: MeshProtos
e: file:///app/plugin-src/ATAK-Plugin-1.0.35/app/src/main/java/com/geeksville/mesh/NodeInfo.kt:36:20 Unresolved reference: MeshProtos
e: file:///app/plugin-src/ATAK-Plugin-1.0.35/app/src/main/java/com/geeksville/mesh/NodeInfo.kt:50:28 Unresolved reference: MeshProtos
e: file:///app/plugin-src/ATAK-Plugin-1.0.35/app/src/main/java/com/geeksville/mesh/NodeInfo.kt:76:27 Unresolved reference: MeshProtos
e: file:///app/plugin-src/ATAK-Plugin-1.0.35/app/src/main/java/com/geeksville/mesh/NodeInfo.kt:102:9 Unresolved reference: ConfigProtos
e: file:///app/plugin-src/ATAK-Plugin-1.0.35/app/src/main/java/com/geeksville/mesh/NodeInfo.kt:103:9 Unresolved reference: ConfigProtos
e: file:///app/plugin-src/ATAK-Plugin-1.0.35/app/src/main/java/com/geeksville/mesh/NodeInfo.kt:104:9 Unresolved reference: ConfigProtos
e: file:///app/plugin-src/ATAK-Plugin-1.0.35/app/src/main/java/com/geeksville/mesh/NodeInfo.kt:105:9 Unresolved reference: ConfigProtos
e: file:///app/plugin-src/ATAK-Plugin-1.0.35/app/src/main/java/com/geeksville/mesh/NodeInfo.kt:130:20 Unresolved reference: TelemetryProtos
e: file:///app/plugin-src/ATAK-Plugin-1.0.35/app/src/main/java/com/geeksville/mesh/NodeInfo.kt:217:26 Unresolved reference: ConfigProtos
e: file:///app/plugin-src/ATAK-Plugin-1.0.35/app/src/main/java/com/geeksville/mesh/NodeInfo.kt:218:26 Unresolved reference: ConfigProtos
e: file:///app/plugin-src/ATAK-Plugin-1.0.35/app/src/main/java/com/geeksville/mesh/NodeInfo.kt:219:26 Unresolved reference: ConfigProtos
e: file:///app/plugin-src/ATAK-Plugin-1.0.35/app/src/main/java/com/geeksville/mesh/NodeInfo.kt:220:26 Unresolved reference: ConfigProtos
e: file:///app/plugin-src/ATAK-Plugin-1.0.35/app/src/main/java/com/geeksville/mesh/util/Extensions.kt:4:28 Unresolved reference: ConfigProtos
e: file:///app/plugin-src/ATAK-Plugin-1.0.35/app/src/main/java/com/geeksville/mesh/util/Extensions.kt:25:5 Unresolved reference: ConfigProtos
e: file:///app/plugin-src/ATAK-Plugin-1.0.35/app/src/main/java/com/geeksville/mesh/util/LocationUtils.kt:3:28 Unresolved reference: MeshProtos
e: file:///app/plugin-src/ATAK-Plugin-1.0.35/app/src/main/java/com/geeksville/mesh/util/LocationUtils.kt:198:24 Unresolved reference: MeshProtos
e: file:///app/plugin-src/ATAK-Plugin-1.0.35/app/src/main/java/com/geeksville/mesh/util/LocationUtils.kt:198:48 Unresolved reference: MeshProtos

> Task :app:compileCivReleaseKotlin FAILED
```

niccellular, 2024-11-16:
looks like you didnt download the protos submodule

AI-Guy-800, 2024-11-18:
Thanks! That's solved the problem, closing issue.

closed 2024-11-18 by AI-Guy-800

maahinberiLW, 2025-02-14:
> looks like you didnt download the protos submodule

I'm trying to build locally and running into the same issue. Where do I get the protos submodule, and how do I incorporate it into the Android project?