meshtastic/ATAK-Plugin #39: Messages Not Received in Longfast

MKme, 2025-01-03:
Google play ATAK + Current Meshtastic plugin results in the ability to direct message other clients and receive messages from other nodes in  longfast (confi'd as channel 0) in the All Chat Rooms Chat.  

The issue:

Responding in this All Chat Rooms chat is not received by any node. I would expect the behavior should be that this is just a reply in LongFast as usual. 

Not sure if this is intended behavior as the linked YouTube demo skips a response in the main chat. Ideas?

RichieFrame, 2025-01-05:
The chat room does NOT send out a regular Meshtastic chat message, rather it sends out a TAK chat packet, over the selected channel. It will show up on other clients only in ATAK chat

MKme, 2025-01-06:
That explains it. Thanks very much. This can be closed.

closed 2025-01-06 by MKme