meshtastic/ATAK-Plugin #43: ATAK crashes when sending DP over Meshtastic plugin

alakida, 2025-02-21:
When trying to send data package via Meshtastic plugin, ATAK shows tooltip that it will reboot the meshtastic device  to SHORT_FAST mode and then the app crashes.

Here's the log of the exception
```
02-21 15:22:23.567 10088 10088 D MeshtasticCallback: Config: device {
02-21 15:22:23.567 10088 10088 D MeshtasticCallback: '-->  role: TAK
02-21 15:22:23.567 10088 10088 D MeshtasticCallback: '-->  node_info_broadcast_secs: 86400
02-21 15:22:23.567 10088 10088 D MeshtasticCallback: '-->}
02-21 15:22:23.567 10088 10088 D MeshtasticCallback: '-->position {
02-21 15:22:23.567 10088 10088 D MeshtasticCallback: '-->  position_broadcast_secs: 86400
02-21 15:22:23.567 10088 10088 D MeshtasticCallback: '-->  gps_update_interval: 30
02-21 15:22:23.567 10088 10088 D MeshtasticCallback: '-->  position_flags: 777
02-21 15:22:23.567 10088 10088 D MeshtasticCallback: '-->  broadcast_smart_minimum_distance: 100
02-21 15:22:23.567 10088 10088 D MeshtasticCallback: '-->  broadcast_smart_minimum_interval_secs: 30
02-21 15:22:23.567 10088 10088 D MeshtasticCallback: '-->  gps_mode: ENABLED
02-21 15:22:23.567 10088 10088 D MeshtasticCallback: '-->}
02-21 15:22:23.567 10088 10088 D MeshtasticCallback: '-->power {
02-21 15:22:23.567 10088 10088 D MeshtasticCallback: '-->  wait_bluetooth_secs: 60
02-21 15:22:23.567 10088 10088 D MeshtasticCallback: '-->  sds_secs: 4294967295
02-21 15:22:23.567 10088 10088 D MeshtasticCallback: '-->  ls_secs: 300
02-21 15:22:23.567 10088 10088 D MeshtasticCallback: '-->  min_wake_secs: 10
02-21 15:22:23.567 10088 10088 D MeshtasticCallback: '-->}
02-21 15:22:23.567 10088 10088 D MeshtasticCallback: '-->network {
02-21 15:22:23.567 10088 10088 D MeshtasticCallback: '-->  ntp_server: "meshtastic.pool.ntp.org"
02-21 15:22:23.567 10088 10088 D MeshtasticCallback: '-->}
02-21 15:22:23.567 10088 10088 D MeshtasticCallback: '-->display {
02-21 15:22:23.567 10088 10088 D MeshtasticCallback: '-->  screen_on_secs: 600
02-21 15:22:23.567 10088 10088 D MeshtasticCallback: '-->}
02-21 15:22:23.567 10088 10088 D MeshtasticCallback: '-->lora {
02-21 15:22:23.567 10088 10088 D MeshtasticCallback: '-->  use_preset: true
02-21 15:22:23.567 10088 10088 D MeshtasticCallback: '-->  region: EU_433
02-21 15:22:23.567 10088 10088 D MeshtasticCallback: '-->  hop_limit: 3
02-21 15:22:23.567 10088 10088 D MeshtasticCallback: '-->  tx_enabled: true
02-21 15:22:23.567 10088 10088 D MeshtasticCallback: '-->  tx_power: 12
02-21 15:22:23.567 10088 10088 D MeshtasticCallback: '-->  override_duty_cycle: true
02-21 15:22:23.567 10088 10088 D MeshtasticCallback: '-->  sx126x_rx_boosted_gain: true
02-21 15:22:23.567 10088 10088 D MeshtasticCallback: '-->  ignore_mqtt: true
02-21 15:22:23.567 10088 10088 D MeshtasticCallback: '-->}
02-21 15:22:23.567 10088 10088 D MeshtasticCallback: '-->bluetooth {
02-21 15:22:23.567 10088 10088 D MeshtasticCallback: '-->  enabled: true
02-21 15:22:23.567 10088 10088 D MeshtasticCallback: '-->  fixed_pin: 123456
02-21 15:22:23.567 10088 10088 D MeshtasticCallback: '-->}
02-21 15:22:23.567 10088 10088 D MeshtasticCallback: '-->security {
02-21 15:22:23.567 10088 10088 D MeshtasticCallback: '-->  public_key: // omitted
02-21 15:22:23.567 10088 10088 D MeshtasticCallback: '-->  private_key: // omitted
02-21 15:22:23.567 10088 10088 D MeshtasticCallback: '-->  serial_enabled: true
02-21 15:22:23.567 10088 10088 D MeshtasticCallback: '-->}
02-21 15:22:23.567 10088 10088 D MeshtasticCallback: '-->
02-21 15:22:23.574 10088 11626 E AndroidRuntime: FATAL EXCEPTION: Thread-664
02-21 15:22:23.574 10088 11626 E AndroidRuntime: Process: com.atakmap.app.civ, PID: 10088
02-21 15:22:23.574 10088 11626 E AndroidRuntime: java.lang.ClassCastException: java.lang.String cannot be cast to java.lang.Integer
02-21 15:22:23.574 10088 11626 E AndroidRuntime: 	at android.app.SharedPreferencesImpl.getInt(SharedPreferencesImpl.java:321)
02-21 15:22:23.574 10088 11626 E AndroidRuntime: 	at com.atakmap.android.meshtastic.MeshtasticCallback.lambda$onMissionPackageTaskComplete$0(Unknown Source:9)
02-21 15:22:23.574 10088 11626 E AndroidRuntime: 	at com.atakmap.android.meshtastic.MeshtasticCallback.a(Unknown Source:0)
02-21 15:22:23.574 10088 11626 E AndroidRuntime: 	at atakplugin.Meshtastic.qf.run(Unknown Source:18)
02-21 15:22:23.574 10088 11626 E AndroidRuntime: 	at java.lang.Thread.run(Thread.java:1012)
```

Device: Lilygo TBeam
Firmware version: 2.5.20-4c97351 Beta
Meshtastic version: Android v2.5.16
Meshtastic ATAK Plugin: v1.0.6

RichieFrame, 2025-02-23:
If you are actually using plugin version 1.0.6, upgrade first as that is a year old and 30 releases behind and does not properly support recent protobufs

alakida, 2025-02-23:
@RichieFrame hey, thanks for the comment. No, I'm using the last release version, 1.0.36, I've probably mistyped the version.
There is a bug in Meshtastic callback when it tries to retrieve the value of the Meshtastic channel from the configuration [right here](https://github.com/meshtastic/ATAK-Plugin/blob/95d9bbad45e56ca139d26d78880915d0528f7c92/app/src/main/java/com/atakmap/android/meshtastic/MeshtasticCallback.java#L86)

I was actually going to create a PR myself but stuck trying to configure the local environment. In particular I can't build the atak-takdev-plugin

niccellular, 2025-03-05:
shoiuld be fixed in 1.0.37

closed 2025-03-05 by niccellular