meshtastic/ATAK-Plugin #45: Changing channel index doesn't work

wiciok, 2025-03-15:
Changing channel index in plugin settings doesn't seem to work on 1.0.37.
I saw this was supposed to be fixed on 1.0.36, but apparently it still sends messages on channel 0.

Versions:
ATAK-Plugin Version: 1.0.37
Meshtastic version: 2.5.16
ATAK version: 5.3.0.10

Repro:
0. Prerequisites: 2 devices (consisting of ATAK EUD and Meshtastic node): A and B
1. Configure 2 devices as follows:
    Device A:
    Channel 0: default LongFast (no encryption)
    Channel 1: LongFast with encryption key X
    Channel 2: LongFast with encryption key Y
    
    Device B:
    Channel 0: default LongFast (no encryption)
    Channel 1: LongFast with encryption key X
    no Channel 2

2. Configure channel index 2 on device A in plugin settings
3. Send message to 'All chat rooms' on device A

Expected result:
Message is not received on device B (as it doesn't have any channel with encryption key Y configured)

Actual result:
Message is received on device B.

niccellular, 2025-03-16:
You're not on WIFI right?

wiciok, 2025-03-16:
No, both WiFi and cellular connection are off.

Also, when sending 'voice memo' messages (which unlike standard GeoChats, are actually visible in Meshtastic app) they are as well on channel 0 Meshtastic chat.

niccellular, 2025-03-16:
> No, both WiFi and cellular connection are off.
> 
> Also, when sending 'voice memo' messages (which unlike standard GeoChats, are actually visible in Meshtastic app) they are as well on Channel 0 in Meshtastic chat.

the voice memo uses the TEXT_MESSAGE_APP portnum for app inter-operability.

niccellular, 2025-03-16:
Please try 1.0.38 and let me know if it resolves the problem.

wiciok, 2025-03-17:
1.0.38 indeed solved the issue. 
Thank you very much for quickly fixing it!

closed 2025-03-17 by wiciok