meshtastic/ATAK-Plugin #54: ATAK 5.4 Crash - NoSuchFieldError

santiman717, 2025-04-23:
Device S23TE  - Android 13 - MR Dated 20 Aug 2024 - 
ATAK:  5.4.0.3
Meshtastic Plugin: 1.1.2
Meshtastick App: 2.5.22

ATAK crashes about 15 seconds into the Meshtastic plugin loading. 

"TAK.error":"java.lang.NoSuchFieldError: No field a of type Lkotlin/Unit; in class Lkotlin/Unit; or its superclasses (declaration of 'kotlin.Unit' appears in base.",

[ATAKCRASH_2025_04_23_22_09_39.json](https://github.com/user-attachments/files/19876387/ATAKCRASH_2025_04_23_22_09_39.json)
[logcat4.txt](https://github.com/user-attachments/files/19876386/logcat4.txt)

Let me know if this isnt' the correct forum to post this. 

Thanks 
BS

RichieFrame, 2025-04-24:
The log is from something else, the actual log is in the json, I have extracted the important part:
```
04-23 15:09:34.926 D/MeshtasticMapComponent(  766): '-->  course: 83
04-23 15:09:34.926 D/MeshtasticMapComponent(  766): '-->}
04-23 15:09:34.926 D/MeshtasticMapComponent(  766): '-->
--------- beginning of crash
04-23 15:09:34.928 E/AndroidRuntime(  766): FATAL EXCEPTION: PublishContactInfoThread
04-23 15:09:34.928 E/AndroidRuntime(  766): Process: com.atakmap.app.civ, PID: 766
04-23 15:09:34.928 E/AndroidRuntime(  766): java.lang.NoSuchFieldError: No field a of type Lkotlin/Unit; in class Lkotlin/Unit; or its superclasses (declaration of 'kotlin.Unit' appears in base.apk!classes2.dex)
04-23 15:09:34.928 E/AndroidRuntime(  766): \tat atakplugin.Meshtastic.Ni$b.invoke(Unknown Source:5)
04-23 15:09:34.928 E/AndroidRuntime(  766): \tat atakplugin.Meshtastic.Ni.d(Unknown Source:43)
04-23 15:09:34.928 E/AndroidRuntime(  766): \tat atakplugin.Meshtastic.Ni.e(Unknown Source:6)
04-23 15:09:34.928 E/AndroidRuntime(  766): \tat atakplugin.Meshtastic.o4$a.a(Unknown Source:58)
04-23 15:09:34.928 E/AndroidRuntime(  766): \tat atakplugin.Meshtastic.o4$a.invoke(Unknown Source:2)
04-23 15:09:34.928 E/AndroidRuntime(  766): \tat atakplugin.Meshtastic.Ni.d(Unknown Source:43)
04-23 15:09:34.928 E/AndroidRuntime(  766): \tat atakplugin.Meshtastic.o4.<init>(Unknown Source:25)
04-23 15:09:34.928 E/AndroidRuntime(  766): \tat com.geeksville.mesh.DataPacket.<clinit>(Unknown Source:38)
04-23 15:09:34.928 E/AndroidRuntime(  766): \tat com.atakmap.android.meshtastic.MeshtasticMapComponent.a(Unknown Source:822)
04-23 15:09:34.928 E/AndroidRuntime(  766): \tat com.atakmap.comms.CommsMapComponent.a(SourceFile:221)
04-23 15:09:34.928 E/AndroidRuntime(  766): \tat com.atakmap.comms.e.a(SourceFile:16)
04-23 15:09:34.928 E/AndroidRuntime(  766): \tat com.atakmap.comms.e.a(SourceFile:7)
04-23 15:09:34.928 E/AndroidRuntime(  766): \tat com.atakmap.comms.e.a(SourceFile:3)
04-23 15:09:34.928 E/AndroidRuntime(  766): \tat com.atakmap.android.cot.CotMapComponent.b(SourceFile:39)
04-23 15:09:34.928 E/AndroidRuntime(  766): \tat com.atakmap.android.cot.CotMapComponent.a(SourceFile:176)
04-23 15:09:34.928 E/AndroidRuntime(  766): \tat com.atakmap.comms.ReportingRate.a(SourceFile:42)
04-23 15:09:34.928 E/AndroidRuntime(  766): \tat com.atakmap.comms.ReportingRate.e(SourceFile:26)
04-23 15:09:34.928 E/AndroidRuntime(  766): \tat com.atakmap.comms.ReportingRate.a(SourceFile:1)
04-23 15:09:34.928 E/AndroidRuntime(  766): \tat com.atakmap.comms.ReportingRate$1.run(SourceFile:3)
04-23 15:09:34.928 E/AndroidRuntime(  766): \tat java.util.TimerThread.mainLoop(Timer.java:573)
04-23 15:09:34.928 E/AndroidRuntime(  766): \tat java.util.TimerThread.run(Timer.java:523)
04-23 15:09:34.930 D/CommsMapComponent(  766): shutting down commo due to crash
```

niccellular, 2025-04-25:
5.4 seems to require a clean install not a upgrade. Try wiping everything and starting fresh.

closed 2025-04-25 by niccellular

santiman717, 2025-04-30:
I'm still seeing crashes even with a fresh install. I deleted ATAK Meshtastic and the Plugin. 

[ATAKCRASH_2025_04_30_23_39_05.json](https://github.com/user-attachments/files/19985839/ATAKCRASH_2025_04_30_23_39_05.json)