meshtastic/ATAK-Plugin #56: ATAK 5.4 Crash - NoSuchFieldError (cont.)

santiman717, 2025-05-01:
I'm still seeing crashes even with a fresh install. I deleted, ATAK, Meshtastic, and the meshtastic plugin. 

[ATAKCRASH_2025_04_23_22_09_39.json](https://github.com/user-attachments/files/19997303/ATAKCRASH_2025_04_23_22_09_39.json)
[ATAKCRASH_2025_04_30_23_39_05.json](https://github.com/user-attachments/files/19997305/ATAKCRASH_2025_04_30_23_39_05.json)
[ATAKCRASH_2025_05_01_00_22_47.json](https://github.com/user-attachments/files/19997304/ATAKCRASH_2025_05_01_00_22_47.json)
[logcat0.txt](https://github.com/user-attachments/files/19997307/logcat0.txt)
[logcat1.txt](https://github.com/user-attachments/files/19997308/logcat1.txt)
[network-1746058886567-log.csv](https://github.com/user-attachments/files/19997306/network-1746058886567-log.csv)
[network-1746058918714-log.csv](https://github.com/user-attachments/files/19997309/network-1746058918714-log.csv)

demizonik, 2025-05-23:
Still happens on CIV version too, same issue

santiman717, 2025-05-23:
Thanks for the bump. I had to go back to ATAK 4.10 to show the capability.

niccellular, 2025-05-27:
Downgrade ATAK to 5.3 or lower.

robert-tessonics, 2025-05-27:
@niccellular do you recommend version 1.1.2 or another build to go along with 5.3? Thanks.