meshtastic/ATAK-Plugin #60: [Feature request] Plugin side de-duplication of Wi-Fi chat packets

Nate-Boyette, 2025-06-11:
Would like to use Meshtastic as a fallback for Wi-Fi based mesh nodes. Python CLI running on the radio works, but is not as fully featured as the official plugin. Only issue I'm running into is chat messages in the "All chat rooms" are duplicated for units that have both Wi-Fi and Meshtastic connection.
With the plugin running directly in ATAK there's no opportunity to handle de-duplication on the Wi-Fi radio side. 

ATAK's normal de-duplication mechanism doesn't work due to the Meshtastic protobuf not preserving the UID of the COT packet.
POI markers do preserve the UID and the position updates are a non-issue in my opinion as you just get slightly more frequent  updates.

Tentative thought on a solution is parsing the CoT messages coming over via Wi-Fi at the multicast port. Add the call sign and chat string to a short term dictionary (like 30 seconds or something) and drop any matching packets coming over from Meshtastic as the Wi-Fi, if it has a connection, should always be the first to arrive.

niccellular, 2025-07-07:
Thank you for posting, I will need to think about how to accomplish this

niccellular, 2025-12-15:
This should be fixed in the new chunking code, all markers are now just EXI compressed and sent, so matching UIDs should prevent duplicates.

closed 2025-12-15 by niccellular