meshtastic/ATAK-Plugin #62: ATAK 5.4.0.9 is crashing after installing Meshtastic plugin 1.1.2

igorsamson, 2025-06-13:
Hello.

I am trying to use the 5.4.0.9 (24009d04) ATAK app with the Meshtastic Plugin version 1.1.2 (5aa55ce2), but it fails immediately after the Meshtastic Plugin is loaded, and the small icon on the main screen turns green.

I have checked the logs, and they look like:
[ATAKCRASH_2025_06_13_15_26_26.json](https://github.com/user-attachments/files/20729190/ATAKCRASH_2025_06_13_15_26_26.json)

I am using an  Android device with the latest (for this device) version RKQ1.200826.002

Any advice?

niccellular, 2025-06-17:
Don't use ATAK 5.4?

santiman717, 2025-06-18:
I haven not been able to successfully use 5.4, I had to roll back to 4.10 to get it to work.

niccellular, 2025-06-19:
5.3 should work too, 5.4 they changed something and I've not had time to fix it.

niccellular, 2025-07-07:
@Jkjvinn 

Please try 

https://github.com/meshtastic/ATAK-Plugin/releases/tag/1.1.14

bisgroves, 2025-07-10:
@amoshalava  can you provide the ATAKCRASH files from the atak/support/logs directory?      

developer note for 5.4:

configurations.all {
    resolutionStrategy.eachDependency { details ->
        if (details.requested.group == 'androidx.core') {
            details.useVersion "1.15.0"
        }
        if (details.requested.group == 'androidx.lifecycle') {
            details.useVersion "2.8.7"
        }
        if (details.requested.group == 'androidx.fragment') {
            details.useVersion "1.8.5"
        }
        if (details.requested.group == 'com.squareup.okhttp3') {
            details.useVersion "4.11.0"
        }
        if (details.requested.group == 'com.squareup.okio') {
            details.useVersion "3.2.0"
        }

    }
}

configurations.implementation {
    exclude group: 'androidx.core', module: 'core-ktx'
    exclude group: 'androidx.core', module: 'core'
    exclude group: 'androidx.fragment', module: 'fragment'
    exclude group: 'androidx.lifecycle', module: 'lifecycle'
    exclude group: 'androidx.lifecycle', module: 'lifecycle-process'
    exclude group: 'com.squareup.okhttp3', module: 'okhttp'
    exclude group: 'com.squareup.okio', module: 'okio'
}