meshtastic/ATAK-Plugin #65: Duplicate nodes/Duplicate messages

CGoosti, 2025-07-14:
Having some issues with getting duplicate messages. Additionally, I'm seeing two nodes for each callsign, as if its displaying the mesh device, and the EDU location. They are close enough to each other, which leads me to believe that's the issue... it also makes the chatting in atak confusing because there is 2 threads for each callsign. See attached for visual.  

Certainly the issue is something I've unknowingly introduced, as i don't fully understand the setting configurations entirely. I suspect its the settings for using ext GPS and internal as a backup... but I don't know for sure.  

Thank you

I follow the instructions in this video for configuration:
https://youtu.be/_v11m2FQQZU?si=Ht-XytxM5heLqogT

[image]

CGoosti, 2025-07-14:
Here you can see the duplicate messages issue, the times get messed up also, so they don't come in order.  

[image]

niccellular, 2025-08-26:
dup of issue #60